mikaelpatel/Arduino-Shell #3: Allow separate input and output streams

mikaelpatel, 2016-02-26:


mikaelpatel, 2016-03-04:
Arduino Stream is input/output. The Print interface is output. This makes it difficult to achieve both input only.

closed 2016-03-04 by mikaelpatel

dpharris, 2016-03-04:
What were you hoping to achieve?

Many mpus have multiple serial interfaces.  I would think applications
might want to send and receive on one of those, but direct errors to the
default serial.  This may be a use for trap functions, or builtin ones: `.
`v `k `m etc to use alternate serial or even `.n `vn `kn `mn to use nth
serial interface.

I think it would be good to make those optional.

mikaelpatel, 2016-03-04:
The idea behind this was to create stream handling a bit like C/C++ with stdin, stdout and stderr. And allow binding to different Arduino Stream classes. 

This has been put on the to-do-later list as there is too little usage to go on right now.

dpharris, 2016-03-04:
Agree.  Should be optional addition.