mikaelpatel/Arduino-Shell #6: Allow scripts in PROGMEM and/or EEMEM

mikaelpatel, 2016-02-26:


dpharris, 2016-02-27:
This would seem to be a good extension.

mikaelpatel, 2016-03-02:
Some example code in [ShellScript](https://github.com/mikaelpatel/Arduino-Shell/blob/master/examples/ShellScript/ShellScript.ino). Uses a buffer and copies scripts from PROGMEM to the buffer before calling the shell. The pattern can be used to reduce SRAM for scripts. Only the command line buffer is needed.

mikaelpatel, 2016-03-07:
Allowing scripts in program memory is an interesting challenge. One solution is to synthesis a combined address space so that the virtual machine can detect if the script is in random access or program memory. Please see commit https://github.com/mikaelpatel/Arduino-Shell/commit/f6f471b66080d9801be6d946c4cbfae0270ff9b7.

dpharris, 2016-03-07:
Nice!

closed 2016-03-08 by mikaelpatel