mikaelpatel/Arduino-Shell #7: Add support for SPI

mikaelpatel, 2016-02-26:
Start with basic SPI transfer function. Suggest opcode(X): X ( data0 -- data1 ).

dpharris, 2016-02-27:
It occurs that you are running out of single characters. (So far you have done an admirable job of assigning them.)  '$' is not taken, but perhaps you need to use an escape character.  Possibly $ could represent 'serial', so $S might be SPI, $I be I2C, $1 be Serial 1, etc.  Also possibly $> and $< could be serial in and out.

dpharris, 2016-02-27:
It occurs to me that these buses are bidirectional, so ine needs two commands each.  Perhaps noncapitalized and capitalized letters?  Eg $s and $S for SPI.  

( You have done such a good job so far, please take these as off-the-top suggestions.)

mikaelpatel, 2016-02-27:
Using $ as a prefix to extend the opcode is a great idea. I have used that before in the vfm project. That is an advanced forth virtual machine. Unfortunately not yet ported to the Arduino. The Shell is a short-cut :). Anyway a virtual member function that is called for $ prefix opcodes is a neat way to extend the Shell opcodes. That goes on the todo list.

mikaelpatel, 2016-02-28:
Added a simple sketch to demonstrate how to handle extended instructions (trap). Please see commit https://github.com/mikaelpatel/Arduino-Shell/commit/430b91e0014c627fcf116bc5a6978e3121be40c2.

dpharris, 2016-02-28:
So, SPI write might be:
   [ 1,2,3,4] $S 
Leaving 4 read values and n on stack?
     5 6 7 8 4

Xn...X0 n --- Yn...Y0 n

closed 2016-02-29 by mikaelpatel