mikaelpatel/Arduino-Shell #8: Add support for TWI

mikaelpatel, 2016-02-26:
I2C requires at least two function; twi-read ( n addr -- x1 ... xn) and twi-write ( x1 ... xn n addr -- ). Many driver functions have the form; 
```
reg 1 addr twi-write n addr twi-read // read a register
val reg 2 addr twi-write // write a register
```

dpharris, 2016-02-28:
Using stack markes is elegant:
[_reg] _addr $I 
2 _addr $i
[23 _reg] _addr $I
[2,3,4,5] _addr $I
(I used $I for twi-write and $i for twi-read)

mikaelpatel, 2016-02-28:
Yes, the stack marker and operations are great contribution by PostScript. There are more very good ideas to be fetch from there. 

I have changed the extended instruction set handling (trap) to a general trap on any op-code that was not handled.

closed 2016-02-29 by mikaelpatel