mikaelpatel/Arduino-Shell #14: General comments

dpharris, 2016-02-27:
Great work.  I have often played with doing something like this. 
Can you add a comment on how to use variables?  
Do you have a compiler / decompiler to go between the expanded forms to compressed forms?

mikaelpatel, 2016-02-27:
> Can you add a comment on how to use variables?
Quick comment here. I will add to the README later. 

The variables are a vector of int's (0..VEC_MAX-1). The address to fetch and store are the index in that vector. 

> Do you have a compiler / decompiler to go between the expanded forms to compressed forms?
No. Nothing I would like to publish yet. This should actually be on the "issue" list.

dpharris, 2016-02-27:
Why is there two 'no ops' chars: ',' and 'N'?

mikaelpatel, 2016-02-27:
',' is for compacting number sequences and N is to hide handling of carriage-return :). There are actually more such as white-space.

closed 2016-02-28 by mikaelpatel