mikaelpatel/Arduino-Shell #17: HEX

dpharris, 2016-02-27:
How can hex be entered?  Suggestion might use X:  X5ef,Xa5+.

(Would have preferred x, and X for execute.)

mikaelpatel, 2016-02-27:
Right now it is only decimal format but I planned to add the traditional C-style with 0x and 0b. This increases the foot-print but will be needed. Also I would like the cell width to be configurable (int as now or long).

mikaelpatel, 2016-02-27:
Please see commit https://github.com/mikaelpatel/Arduino-Shell/commit/8ba71a0e052fce20e2940eb910345b819f2ee919.

dpharris, 2016-02-27:
Yes, just saw it in the code!

mikaelpatel, 2016-02-27:
The current implementation has a few short-cuts. It is possible to give 0b12345, i.e. binary base does not check the character set. I will get back to this later on.

mikaelpatel, 2016-02-27:
Hexadecimal should be working fine so I will close this issue.

closed 2016-02-27 by mikaelpatel

mikaelpatel, 2016-02-27:
Commit https://github.com/mikaelpatel/Arduino-Shell/commit/2f19d35a2caea6c5ac62a84b2b476b8c7f4f3b9d fixes the issue with accepting illegal numbers.