mikaelpatel/Arduino-Shell #18: Case statement

dpharris, 2016-02-28:
This would be useful.  Possible do it now, but messy:
   2@  d0={(1st)}i d1={(2nd)i 2d={(3rd)}i

Alternately could use stack marker and 'C' perhaps:

 [{(1st)}{(2nd)}{(3rd)}{(4th)}]2@C

C would need to check selector range, execute the appropriate block, and delete the block addresses from the stack.  

May be guilding the lily.

mikaelpatel, 2016-02-28:
Avoided this as it indirectly requires handling of multi-line (or larger command line buffer). The best solution for dispatch is to use several variables and index on their address.

dpharris, 2016-02-28:
Great idea!  So:  if my state is in var 6, then:
{...}\a0!  {...}\a1!  {...}\a2!  {...}\a3!   6@@x

mikaelpatel, 2016-02-28:
Yes! This is a great way to implement state-machines (see Cosa/FSM). I will close this issue for now. Might get back to it later.

closed 2016-02-28 by mikaelpatel