mikaelpatel/Arduino-Shell #19: Stack markers

dpharris, 2016-02-28:
I am not sure I fully understand stack markers [].
''"
[6 34$I]
'''
would read 6 bytes from I2C device 34, and leaves 6 items on the stack with a count 6 on top, useful for future processing.  

How does that differ from just:
'''
6 34 $I 6
'''
?
I will go and reread PS (last time 25 yrs ago!).

mikaelpatel, 2016-02-28:
No difference at all until you start using functions with variable number of parameters or return values.

mikaelpatel, 2016-02-28:
PS syntax for array is [...]. Behind this is the functions **mark**, **counttomark** and allocation of the array. I have "used" [ as mark, and ] as counttomark. A simplification for stack based vectors (and with the new stack direction, from top stack to bottom in memory).

closed 2016-02-29 by mikaelpatel