mikaelpatel/Arduino-Shell #21: Escape

dpharris, 2016-02-29:
Can you explain how you saw '$' working?  

I had assumed that $ sucked in the next char and sent that to the trap(char) function.  I messed with the code and got that to work.  However, ...

It seems like you are trying for a more general mechanism that just traps to a trap(char) function, and then the trap function can do anything (within reason).  Is it allowed to play with the stream, i.e. peek and poke it?

mikaelpatel, 2016-02-29:
There is no escape/prefix op code. Any op code that is not used can be used. And the trap function can use multiple op codes. The vision is multiple co-processors. The Arduino instructions would move out of the inner interpreter and into a trap handler. This is for the next step when the code get ported back to linux.

closed 2016-02-29 by mikaelpatel

dpharris, 2016-02-29:
Ok, I am starting to grok this ... nice idea.  Perhaps one token should be
reserved for an escape, though, so one can always add something, event if
all other tokens are used.  It also lets one standardize somewhat, since
its fixed.