mikaelpatel/Arduino-Shell #25: timeout

dpharris, 2016-03-03:
I see a new t(timeout) token.  I need this in my app!  
I have been using: 
  if(MIllis()>next) { ....  next+=period; }  // this is slightly faster, and gives consistent period

[[ roll-over is a problem, and more of a problem with Shell, since M is only a word.  
   The math should be something like:
       if( (uint16_t)(M-next)>0x3FFF ) ...   when using uint16_t  (I think)
  Or
     if(  (int16_t)(M-next)>0 ) ... if using signed arithmetic (I think) 
]]

mikaelpatel, 2016-03-03:
Don't really understand this issue. Did you find a bug in the current code or was this a comment. 
The proposed rewrites do not work. Actually non of them. The first has a initialization problem. Did you test any of these and is there an actual performance gain in the interpreter (the overhead before coming to the operation code implementation is high)?

dpharris, 2016-03-03:
Ignore me.  Your math works perfectly.  Thanks!

mikaelpatel, 2016-03-03:
I think you are on to something but I could not get it to work with all the mapping from uint32_t to unsigned, etc. Might get back to this later on. The current version is not "pretty" but it works.

mikaelpatel, 2016-03-03:
"t" for "?timeout" is renamed to "E" for "?expired". This is mainly motivated by uppercase used to Arduino extensions of the virtual machine. Please see commit https://github.com/mikaelpatel/Arduino-Shell/commit/ba8a5ce9e56fcc07ee97e6d0b4f8f3e0bb8db088.

closed 2016-03-04 by mikaelpatel