mikaelpatel/Arduino-Shell #28: Improve operation codes

mikaelpatel, 2016-03-04:
The operation codes for frames, parameters, variable lookup and escape could be more logical.

1. "n$..-n$" could be "n\\...-n\".
2. "n_" could then be "n$" which is RPN of shell parameters.
3. "\name" would have to change to for instance "`name".
4. "`" would change to "_".

The old syntax:
```
{2$2_@1_@-2};\swap!
3,4\swap:
```
would become:
```
{2\2$@1$-2\};`swap!
3,4`swap:
```

mikaelpatel, 2016-03-04:
Please see commit https://github.com/mikaelpatel/Arduino-Shell/commit/443398753d03ccc3b1d86c7badd89f67ec0436fe.

dpharris, 2016-03-04:
> Please see commit 4433987

!   Can you call trap for any name not recognized?   That argues for the
:name and name:, say.

dpharris, 2016-03-04:
... but that doesnt work, sorry, needs to be a prefix for the call.

mikaelpatel, 2016-03-04:
This is just an attempt to clean up the operation code (character) assignments and make the frame and parameter more logical.

mikaelpatel, 2016-03-04:
Some additional renaming; "b" => "c", "k" <=> "K", "y" => "Y". Please see commit https://github.com/mikaelpatel/Arduino-Shell/commit/c7b61e1b74fbe161a14b4d422983b8102b15a429.

closed 2016-03-04 by mikaelpatel