mikaelpatel/Arduino-Shell #32: Define named scripts in sketch (C++)

mikaelpatel, 2016-03-08:
It should be possible to set a variable or function from the sketch. This allows the application to define a set of scripts.

    shell.set(name, value);
    shell.set(name, script);

mikaelpatel, 2016-03-08:
Please see commit https://github.com/mikaelpatel/Arduino-Shell/commit/e6f910e5ef395a626d02e8af9dd5c7128099c5ff.

dpharris, 2016-03-09:
How about loading the stack before running the script?  
Can I do: 
   shell.push(0); shell.push(4); shell.execute(script0);

mikaelpatel, 2016-03-09:
That has been available for a while now :).

mikaelpatel, 2016-03-10:
Allow scripts from SRAM and PROGMEM to be set. Please see commit https://github.com/mikaelpatel/Arduino-Shell/commit/dfa2fc5d89ec6461a474555ed1942681c839e35a.

closed 2016-03-11 by mikaelpatel