mikaelpatel/Arduino-Shell #39: SIMPL

dpharris, 2016-03-15:
I noticed on the 43oh group that forth has sprung forth in several forms.  The SIMPL language mentioned here: http://forum.43oh.com/topic/9429-simpl-a-tiny-language-for-msp430/ looks similar to your efforts.  I haven't analyzed in detail, but I like yours!

mikaelpatel, 2016-03-15:
Thanks for that reference. I have not seen that before. It has very much in common with Shell but there are also big differences. It will be interesting to compare when that project is completed.

dpharris, 2016-03-15:
They suggest that there is a family of such languages.  At this point Shell
is a pretty complete system.  A comparison will be interesting. I had
planned on using Shell on AVR and ARM boards such as the Tiva, Teensy, and
Nucleo boards.

mikaelpatel, 2016-03-15:
The family of token thread interpreter is large. The most famous is maybe OpenBoot which is designed by Mitch Bradley, former engineer at Sun Microsystems. There is even an IEEE standard on this. 

I think the most important and novel aspect is using characters and using them as more or less one letter abbreviations of the stack machine operations in Forth/Postscript. This set of character can be seen as the "core" set. In Shell this is symbol characters plus lower case letters. 

There is a difference in the ability to handle the memory variants and address space of MCUs. These have typically RAM, EEPROM and Program Memory. Each in a separate address space. As the amount of RAM is very limited it cannot be used for long term storage of scripts.

dpharris, 2016-03-15:
Yes, I have had a long history with Forth predating Mitch's work on
bootloading.   I am glad it was opened up.

mikaelpatel, 2016-03-15:
I met Mitch a couple of times at the Rochester Forth conferences (late 80'ies). Presented some research papers at the conference. You can find them on my LinkedIn page. One is on debugging/development support environments and the other on modularization and supporting "programming-in-the-large".

dpharris, 2016-03-15:
Neat.  I am a physician, so my interest is non-professional, although I
have a CSc undergrad degree, and did programming related to my PhD
research.   Computer language design has been a longstanding interest -
particularly Self and Smalltalk.  I am hacking around C++ and Java code.

As I recall Mitch was very active in Forth community before OpenBoot ...
but that project certainly raised Forth's profile!