mikaelpatel/Arduino-Shell #53: Yo

dpharris, 2016-07-15:
Sorry .. Got busy and lost you for a bit.   Anything new?  
David

mikaelpatel, 2016-07-25:
Sorry for the delay; quick answer is no. Might be time for a rewrite :)

mikaelpatel, 2016-12-21:
@dpharris The answer is now yes. Started a rewrite towards a more traditional forth kernel (port of VFM). Please see https://github.com/mikaelpatel/Arduino-FVM. Merry Christmas and Happy New Year.

dpharris, 2016-12-21:
Hi

Long-time no-hear.  Many happy felicitations!

Can you summarize?  Did we lose the character-encoding?

Cheers
David

mikaelpatel, 2016-12-21:
Arduino-FVM is more a traditional token threaded Forth kernel. The instruction encoding is not set yet but as there are several instructions with offsets and data in the instruction sequence a simple compiler needed. Also currently "scripts" can only be executed from program memory. The memory architecture is not yet abstracted. 

The goals are somewhat different; low foot-print, high performance, multi-tasking and easier to integrate with C++. 

The kernel with approx 100 primitives takes up 3Kbyte, and the threading overhead is 1-2 us. The delay primitive uses yield and a forth task can be resumed from the sketch. This allows easy and powerful integration patterns. Some measurements can be found in the Blink example sketch; https://github.com/mikaelpatel/Arduino-FVM/blob/master/examples/Blink/Blink.ino

I hope that some of the ideas and solutions will ripple back to the Arduino-Shell and help it evolve. 

Cheers, Mikael.

dpharris, 2016-12-21:
> master/examples/Blink/Blink.ino

Thank you, that is a good introduction.   Looking at the blink sketch is
now helpful and understandable.
Good work,
cheers,
David

mikaelpatel, 2017-01-12:
@dpharris The FVM is starting to have more of the necessary "parts". There are more example sketches and a token compiler has been added. 

At first it might be strange to use an Arduino as a token compiler. It basically reads Forth and writes C++ definitions for the virtual machine (source code to include in sketch). The Test sketch token code is generated with the compiler. 

The latest update to the trace allows indentation of the call depth. Together with task reference this gives a very enlightening printout of what is going on in the virtual machine. Cheers! Mikael

mikaelpatel, 2017-01-23:
@dpharris A Forth style outer interpreter is now available as an example sketch in the FVM project. It will allow definitions in both data and program memory. And has much of the flavors from Arduino-Shell project. The kernel dictionary can be reduced to single character strings alternatively the tokens may be mapped to single characters to allow. 

An interesting optimization in the FVM inner interpreter is tail call reduction, i.e. return address are pushed on calls when the succeeding token is not EXIT, the call becomes a jump instead. This removes unnecessary return stack push and pop, and calling EXIT.  For more details see FVM_KERNEL_OPT, https://github.com/mikaelpatel/Arduino-FVM/blob/master/FVM.cpp#L156.

dpharris, 2017-05-07:
> master/FVM.cpp#L156.

Hi Mikael --

I am just restarting to get my head around your developments.  BTW, the
'screenshots' on github are not working for me.

David

dpharris, 2017-12-24:
Hi again!   Thinking again of applying your forths.  Will need to review them again.  I would need to offset the eeprom pointer to avoid my use of it already.  Code to flash sounds very good.  Merry Christmas!